Tom-evnut/SimpBMS #4: Manual is unclear on "7 – Battery Capacity " setting.1

raphael303, 2022-11-21:
The manual states:
"7 – Battery Capacity = each group of parallel cells monitored per slave capacity in Ah"

To me it is unclear what to enter here. The capacity of one module, for example 210Ah for a Tesla module, or 420Ah, if I have two in parallel. I don't know what "each group" means in this context. Does it mean "module". Why is it sometimes "group of cells" and sometimes "modules" in the manual?

"Battery Capacity – Capacity of parallel cells monitored by BMS Slave"
The cells within one module, or all the parallel cells aggregated over "n" paralleled modules?

Somewhere else the manual says:
"0 – slaves in parallel = amount of slave strings in pack (will multiply total pack capacity)"
Which implies it might be 210Ah, which I have to enter, and 420 will be calculated automatically. 

To me, this is not conclusive or at least confusing. 
I set my battery capacity setting to 420Ah (2 Tesla modules parallel)
If I should set it to 210Ah, imho, the setting should be named "module capacity" or "cell group capacity".

Tom-evnut, 2022-11-26:
Sure, but read the part on page 20 which explains this. Naming it other things wont solve it for everyone that is why the page is created. 

It is Capacity per module.

closed 2022-11-26 by Tom-evnut

raphael303, 2022-11-27:
Thank you for clearing that up.
Page 20 I had quoted above in OP:
"Battery Capacity – Capacity of parallel cells monitored by BMS Slave"

The parallel cells could be over all modules or within one module. Nowhere it states clearly what is ment by it. It might seem obvious to you, but to me, and the the seller of SimpBMS this does not clear it up the way you have in your post.

I'm paraphrasing what you essentially said:
<"Battery Capacity" as you find it in the settings dialog is in fact "capacity per module">

It's your device. It's a great one, great engineering, really, and it's up to you to name anything whatever you want.

but, for the record only. After your explanation I know what to set, but don't get why not call it "capacity per module" in the settings instead of battery capacity. As "battery" really is the whole collective of modules. If "battery" is synonymous with "module" I don't see the point for the different words anyhow.

But, since it is now settled here, I hope that anybody with the same problems will find this exchange with a google search. 

Thank you anyway for your great work! It's really a very smart and capable device. Imagine how many tons of CO2 have been kept out of the atmosphere thanks to you having given so many people access to second-life batteries. Speaking of compensation-efforts, you have certainly done your deed! :)